mshr-h/vscode-verilog-hdl-support #418: [BUG] cant run verilator under WSL

zoomer-k, 2023-05-30:
**Describe the bug**
cant run verilator under WSL

**Environment (please complete the following information):**
 - OS: [e.g. Windows 10 22H2 build 19045.2965]
 - VS Code version [1.78.2]
 - Extension version [1.9.0 - latest]

**Steps to reproduce**
Steps to reproduce the behavior:
1. Open .v or .sv file
2. Check output (verilog)
3. no warning detect

**Log**
Attach the log generated while following the above steps
2023-05-30 11:56:14.235 [info] mshr-h.veriloghdl is now active.
2023-05-30 11:56:14.235 [info] [Ctags] ctags manager configure
2023-05-30 11:56:14.235 [info] [lint-manager] Using linter: verilator
2023-05-30 11:56:14.235 [info] Rewrote docUri to /mnt/d/verilog/test.sv for WSL
2023-05-30 11:56:14.235 [info] Rewrote docFolder to /mnt/d/verilog for WSL
2023-05-30 11:56:14.235 [info] [verilator] Execute
2023-05-30 11:56:14.235 [info] [verilator]   command: wsl verilator -sv --lint-only -I"/mnt/d/verilog"  "/mnt/d/verilog/test.sv"
2023-05-30 11:56:14.235 [info] [verilator]   cwd    : /mnt/d/verilog
2023-05-30 11:56:14.235 [info] mshr-h.veriloghdl activation finished.
2023-05-30 11:56:14.235 [info] [verilator] 0 errors/warnings returned

**Expected behavior**
some warning detect
(ver 1.8.1 can detect warning)

**problem point**
ver 1.9.0
```typescript
docFolder = this.convertToWslPath(docFolder);
this.logger.info(`Rewrote docFolder to ${docFolder} for WSL`);
```
docFolder is wls path .
but child.exec( { cwd : cwd }) need windows path 

**solve way**
add step cwd select (isWindows)
```typescript
  let cwd: string = this.runAtFileLocation
    ? isWindows
      ? cwdWin
      : docFolder
    : vscode.workspace.workspaceFolders[0].uri.fsPath;
```
cwdWin is
```typescript
let cwdWin: string = path.dirname(docUri);
```

I checked solving problem but not check side effect

mshr-h, 2023-05-30:
Ok. Looks good to me.
Can you make the PR?

zoomer-k, 2023-05-30:
thanks your attention
I'll try it
fix SlagLinter.ts same time

commit 66482256 (2023-06-01): fix #418  slang and verilator  cant run under WSL (#419)

* Update VerilatorLinter.ts

add cwdWin for windows

* Update SlangLinter.ts

add cwdWin for windows

* Update CHANGELOG.md

add unreleased log

* minor update

---------
commit c35b4419 (2023-06-06): fix #418 again
closed 2023-06-01 by mshr-h