mshr-h/vscode-verilog-hdl-support #432: Proposal to update description of ctags in README.md

hirooih, 2023-07-11:
Many of the same questions and requests related to ctags have been registered in issues, #66, #84, #130, #142, and #151.

### Proposal 1

Add the following [your comment](https://github.com/mshr-h/vscode-verilog-hdl-support/issues/66#issuecomment-481743325) in [ctags-integration](https://github.com/mshr-h/vscode-verilog-hdl-support#ctags-integration) section of `README.md`.

> Currently the extension supports only tags for one file (the currently open file) and not the entire work space.

### Proposal 2

Introduce extensions for ctags to support tags for any files.

For example [ctags companion extension](https://marketplace.visualstudio.com/items?itemName=gediminaszlatkus.ctags-companion) works great with this extension as I commented [in Dec, 2020](https://github.com/mshr-h/vscode-verilog-hdl-support/issues/66#issuecomment-748618853) 

Just add the following settings on `.vscode/settings.json` in your workspace.

```json
{
    "ctags-companion.command": "ctags -R --fields=+nKz -f .vscode/.tags --langmap=SystemVerilog:+.v -R rtl /opt/uvm-1.2/src",
    "ctags-companion.readtagsEnabled": true,
}
```

It supports files not only in the workspace, but also outside the workspace. (ex. `/opt/uvm-1.2/src` in the example above)

It also recently added support for the `readtags` command in the Univsersal Ctags distribution, which allows for faster searches even on large projects.

***

Shall I send a PR?

mshr-h, 2023-07-16:
@hirooih 
Thanks for the proposals!
All of them looks good to me.
Ca you make the PR?

hirooih, 2023-07-16:
I've sent a PR.
Thanks.

closed 2023-07-20 by mshr-h