mshr-h/vscode-verilog-hdl-support #446: [Enhancement] Pass arguments to lsp

Yakkhini, 2023-09-05:
**Describe the bug**
I'm using verible-verilog-ls as my SystemVerilog Language Server and I found that there is no options to pass arguments for verible lsp or other language servers. I found that the plugins source code has wrote an interface for passing arguments to language servers in `src/extensions.ts`, function `setupLanguageClient()`, argument `serverArgs` but with some reason it isn't been exposed as an setting option.

I want to implement (or modify) this function to read lsp arguments from settings and pass to language server. Verible lsp working is enough for me, but it seems like modify the function would effects other lsp. So there is any plan to implement this feature or any reason not to implement it?

If the maintainer agree for implementing the feature, I would like contribute to this plugin.

(Plus, pass argument in path straightly not work, at least the verible lsp can't work. I will put relative info below.)

**Environment (please complete the following information):**
 - OS: NixOS 23.05.20230829
 - VS Code version 1.78.2
 - Extension version 1.13.0
 - Language server: verible-verilog-ls

**Steps to reproduce**
Steps to reproduce the behavior:

```json
...
"verilog.languageServer.veribleVerilogLs.path": "verible-verilog-ls --rules=-unpacked-dimensions-range-ordering",
...
```

**Log**
```
[Error - 5:15:18 PM] veribleVerilogLs language server client: couldn't create connection to server.
Launching server using command verible-verilog-ls --rules=-unpacked-dimensions-range-ordering failed. Error: spawn verible-verilog-ls --rules=-unpacked-dimensions-range-ordering ENOENT

```

**Expected behavior**
Verible lsp works with custom rules.

**Actual behavior**

Couldn't create connection to server.

Yakkhini, 2023-09-05:
The command `verible-verilog-ls --rules=-unpacked-dimensions-range-ordering` works well in shell.

bolinfest, 2023-09-05:
Please consider https://github.com/mshr-h/vscode-verilog-hdl-support/issues/447 when adding support for this feature.

bolinfest, 2023-09-05:
@Yakkhini you could seemingly work around this by defining your own Bash script like so:

```
#!bin/bash

verible-verilog-ls --rules=-unpacked-dimensions-range-ordering "$@"
```

and then specifying the path to your Bash script as the value for the `"verilog.languageServer.veribleVerilogLs.path"` VS Code setting.

Yakkhini, 2023-09-06:
> @Yakkhini you could seemingly work around this by defining your own Bash script like so:
>
> ```
> #!bin/bash
>
> verible-verilog-ls --rules=-unpacked-dimensions-range-ordering "$@"
> ```
>
> and then specifying [...]

It seems like a good idea. ;) I will give it a try later. As now, I've decided setting verible language server experimental feature to false and use this plugin with CHIPS ALLIANCE' official plugin together.

Thanks!

closed 2024-08-26 by mshr-h