mshr-h/vscode-verilog-hdl-support #447: [BUG] prefer child_process.execFile() to child_process.exec()

bolinfest, 2023-09-05:
**Describe the bug**
Building up a command string using user-provided arguments and passing it to `child_process.exec()` runs the risk of shell injection. This can be easily avoided by switching from `exec()` to `execFile()`. Example:

https://github.com/mshr-h/vscode-verilog-hdl-support/blob/ca47b91e3fd8ab523b275386786c5b352caa4a59/src/linter/ModelsimLinter.ts#L55-L64

would be safer as:

```
    var process: child.ChildProcess = child.execFile(
      this.modelsimPath,
      [
        'vlog',
        '-nologo',
        '-work',
        this.modelsimWork,
        doc.fileName,
        this.modelsimArgs,
      ],
```

Though admittedly this does not seem *quite* right because it seems like `this.modelsimArgs` can be an arbitrary Bash command. Ideally it would be specified as a list of strings and then you could do:

```
    var process: child.ChildProcess = child.execFile(
      this.modelsimPath,
      [
        'vlog',
        '-nologo',
        '-work',
        this.modelsimWork,
        doc.fileName,
      ] + this.modelsimArgs,
```

**Environment (please complete the following information):**
- does not matter

**Steps to reproduce**
- based on static analysis

**Log**
N/A

**Expected behavior**
In the current implementation, if someone sneaks in something like a `doc.fileName` this is a Bash command, such as `; exit;` or what have you, then that's evidence of a possible Remote Code Execution vulnerability.

**Actual behavior**
Using `execFile()` should prevent the sort of vulnerability described.

**Additional context**
To fix this properly, it seems like some user config options might have to change from `string` to `string[]`, which is likely a breaking change unless you want to continue to support the insecure version.

AndrewNolte, 2024-04-12:
I don't really understand why security is a problem for a vscode extension? On vscode, you have a shell for you own machine anyway?

AndrewNolte, 2024-04-24:
This is fixed on https://github.com/AndrewNolte/vscode-system-verilog