mshr-h/vscode-verilog-hdl-support #472: [BUG] No highlighting for module/instance name

michaelw618, 2024-03-21:
**Describe the bug**
Module and instance name highlighting does not work if parameter assignment starts at the line after module name:

[image]

**Environment (please complete the following information):**
 - OS: Windows 10 Pro 10.0.19045
 - VS Code version 1.87.2
 - Extension version v1.13.2
 - color themes dark

**Steps to reproduce**
use code from screen shot above in a .sv file

**Log**
N/A

**Expected behavior**
I expect that the module name <my_module> appears in blue and the instance name <u_my_module> in light blue/cyan

**Actual behavior**
see above

OmarElfouly, 2024-10-28:
Hello @mshr-h 

I hope all is well on your side :)

I noticed that in systemverilogs case we are using the .yaml to generate the json instead of just the json so i wanted to ask if there was a specific reason we are doing so.

Also I am pretty sure the issue is the use of 
`[ \\t\\r\\n]` 
Which is:
[image]

instead of
`[ \t\r\n]`
Which is
[image]

mshr-h, 2024-10-28:
@OmarElfouly 
The reason we use yaml instead of json is human readability.
PRs are welcome!