mshr-h/vscode-verilog-hdl-support #479: [BUG] Module instantiation doesn't instantiate parameters

toTheSky, 2024-04-10:
**Describe the bug**
Module instantiation instantiates ports but not parameters.

**Environment (please complete the following information):**
 - OS: Win 11
 - VS Code version 1.88.0
 - Extension version 1.13.5
 - ctags version v6.1.0 x64

**Steps to reproduce**
Steps to reproduce the behavior:
1. Run Command Palette
2. Type and run "Verilog: Instantiate Module"
3. Select Verilog of SystemVerilog file
4. See error

**Log**
I don't know where the log is located.

**Expected behavior**
Parameters should be instantiated too.

**Actual behavior**
Only ports are instantiated.

**Additional context**
I tried to add additional arguments to ctags path (`ctags.exe --fields-Verilog=+{parameter}`) but it does't help.

AndrewNolte, 2024-04-12:
This is fixed on https://github.com/AndrewNolte/vscode-system-verilog. It'll also fill in the instantiation as you type `'Module #(`, and it support interfaces

toTheSky, 2024-04-20:
> This is fixed on https://github.com/AndrewNolte/vscode-system-verilog. It'll also fill in the instantiation as you type `'Module #(`, and it support interfaces

Thanks. Can it be merged with the main repo?

RickyTino, 2024-08-12:
Hi, I am quite sure that this problem is caused by #102 between v1.13.0 and v1.13.1. 
The latest version without such problem is v1.13.0.
This commit should probably be reverted?

hirooih, 2024-08-31:
@Raamakrishnan 

Thank you for your fix.

Now I see my fixes of #457 was correct, but I had to fix one more line.

commit 11dd1647 (2024-08-26): Fix #479 to add parameters back into Module instantiation (#496)