mshr-h/vscode-verilog-hdl-support #482: [BUG] iverilog includePath not working

jiang131072, 2024-04-14:
**Describe the bug**
iverilog includePath not working. Not even in IcarusLinter's Execute log.

**Environment (please complete the following information):**
 - OS: Windows1 11 22H2 22621.3447
 - VS Code version 1.88.1
 - Extension version 1.13.5
 - using ctags companion, copilot and the extension

**Steps to reproduce**
settings.json:
```json
	{
    "verilog.linting.linter": "iverilog",
    "verilog.linting.iverilog.runAtFileLocation": true  // This is not a must for reproduce the bug
    "verilog.linting.iverilog.includePath": [
        "../../sources_1/new",
    ],
```
then open a .v file.

**Log**
```log
2024-04-14 05:29:07.781 [info] [LintManager] [IcarusLinter] Executing IcarusLinter.lint()
2024-04-14 05:29:07.781 [info] [LintManager] [IcarusLinter] iverilog binary path: iverilog
2024-04-14 05:29:07.781 [info] [LintManager] [IcarusLinter] Execute
2024-04-14 05:29:07.781 [info] [LintManager] [IcarusLinter]   command: "iverilog -t null -g2005 -I \"e:\\sources_1\\new\"  \"e:\\proj\\dsd1-1\\dsd1-1.srcs\\sim_1\\new\\tb_adder_4bit_lookahead.v\""
2024-04-14 05:29:07.781 [info] [LintManager] [IcarusLinter]   cwd    : "e:\\proj\\dsd1-1\\dsd1-1.srcs\\sim_1\\new"
2024-04-14 05:29:07.788 [info] [VerilogDocumentSymbolProvider] [VerilogSymbol] Symbols Requested: file:///e%3A/proj/dsd1-1/dsd1-1.srcs/sim_1/new/tb_adder_4bit_lookahead.v
2024-04-14 05:29:07.788 [info] indexing "e:\\proj\\dsd1-1\\dsd1-1.srcs\\sim_1\\new\\tb_adder_4bit_lookahead.v"
2024-04-14 05:29:07.788 [info] executing ctags
2024-04-14 05:29:07.789 [info] building symbols
2024-04-14 05:29:07.789 [error] No output from ctags
2024-04-14 05:29:07.789 [info] [VerilogDocumentSymbolProvider] 0 top-level symbols returned
2024-04-14 05:29:07.857 [info] [LintManager] [IcarusLinter] 1 errors/warnings returned
```

no output from ctags is another bug.

**Expected behavior**
Include the path correctly

**Actual behavior**
Nothing happened

Raamakrishnan, 2024-08-25:
> no output from ctags is another bug.

Are you using ctags through snap? If so, can you try using the pre-built binary or building it yourself?
There seems to be an issue with snap not having permissions to access certain files

jiang131072, 2024-08-26:
@Raamakrishnan I am on Windows 11, universal ctags from [winlibs - MinGW](https://github.com/brechtsanders/winlibs_mingw)

Raamakrishnan, 2024-08-31:
My bad not noticing that you are running on Windows. From the log you provided, I think you have not set the path to ctags binary. Can you check what you have in your workspace settings for `verilog.ctags.path` ?