mshr-h/vscode-verilog-hdl-support #484: [BUG] Excessive file access when using Samba

xachb, 2024-04-16:
**The bug**
When using xvlog on windows, and accessing files on Red Hat sever over a Samba, I am pegging the remote servers CPU.

**Environment:**
 - Local OS: Windows 10 Enterprise 22H2 build 19045.4170
 - Server OS: Red Hat Enterprise Linux Server release 7.9 (Maipo)
 - VS Code version: 1.88.1
 - Extension version: 1.13.5
 - Lint tools: xvlog (Vivado 2020.2 on local windows machine) linting

**Steps to reproduce**
Steps to reproduce the behavior:
1. Enable Verilog-HDL/SystemVerilog/Bluespec SystemVerilog extension (with xvlog linter running on windows machine)
2. Run top on my linux server (watch CPU utilization for smbd process peg around 65-95% of cpu useage), file access is slow and sometimes unresponsive
3. Disable extension (watch smdb utilization drop to nothing)

**Expected behavior**
I would expect reasonable file access with this extension, all of my other extensions in vscode are doing just fine.

**Actual behavior**
Blowing up Samba on Red Hat server. Is it possible that the linter is running at the file location, and not in the local workspace?

I really appreciate this linter, but unfortunately, I have to disable it in my environment because it is breaking our remote server, I hope you can find a fix. 

Thank you,
Zach

defvs, 2025-04-09:
The issue is that the extension triggers a #find command which leads to ripgrep doing a project-wide search. That causes massive headaches when the filesystem is slow, such as on SMB.

defvs, 2025-04-09:
I dug more into this - on my side, this is caused by a full project search triggered when hovering a symbol : https://github.com/mshr-h/vscode-verilog-hdl-support/blob/0874ce87e312a2138ab56ebf809ce7c1ed76d1fa/src/providers/HoverProvider.ts#L24

which leads to a search for `**/${symbol}.sv`:
https://github.com/mshr-h/vscode-verilog-hdl-support/blob/0874ce87e312a2138ab56ebf809ce7c1ed76d1fa/src/ctags.ts#L369

If these take a lot of time to finish, they start to pile up and bog the CPU and Filesystem down.

@

defvs, 2025-04-09:
A temporary fix would be to disable hovering, by adding this to your settings.json:
```json
    "[systemverilog]": {
        "editor.hover.enabled": false
    },
```