mshr-h/vscode-verilog-hdl-support #487: [BUG] Verible verilog formatter arguments not working

riuandg5, 2024-06-23:
**Describe the bug**
I changed the default indentation width from 2 to 4 by providing arguments but verilog code is still using 2 as indentation width after formatting.

**Environment (please complete the following information):**
 - OS: Windows 10.0.19045 build 19045]
 - VS Code version 1.90.2
 - Extension version v1.14.1
 - Verible verilog formatter version: v0.0-3704-g1d393d43 Commit 2024-06-18 Built 2024-06-19T06:52:13Z

**Steps to reproduce**
1. My settings:
    ```json
    "[verilog]": {
            "editor.defaultFormatter": "mshr-h.veriloghdl"
    },
    "verilog.linting.linter": "xvlog",
    "verilog.languageServer.veribleVerilogLs.enabled": true,
    "verilog.formatting.verilogHDL.formatter": "verible-verilog-format",
    "verilog.formatting.veribleVerilogFormatter.arguments": "--indentation_spaces=4"
    ```
2. Code after formatting:
    ```verilog
    `timescale 1ns / 1ps
    
    module clk_divider #(
        parameter integer DIV_VALUE = 5
    ) (
        input clk_in,
        output reg clk_out = 0
    );
    
      reg [31:0] count = 0;
    
      always @(posedge clk_in) begin
        if (count == (DIV_VALUE - 1)) begin
          count   <= 0;
          clk_out <= ~clk_out;
        end else begin
          count <= count + 1;
        end
      end
    endmodule
    
    ```

**Log**
I don't know how to get the logs for the above format operation.

**Expected behavior**
Verilog code should be formatted to use indentation width of 4.

**Actual behavior**
Verilog code is being formatted with indentation width of 2.

**Additional context**
When directly using the verible-verilog-format on command line with --indentation_spaces=4 as argument, it is working as expected.

hi0t, 2024-07-31:
I guess that your configuration uses the formatter from veribleVerilogLs. verible-verilog-ls also has the --indentation_spaces=4 setting. But I didn't find a way to pass arguments when starting the ls. When setting
`verilog.languageServer.veribleVerilogLs.path": ".vscode/verible-verilog-ls --indentation_spaces=4`
an error appears:
`[Error - 11:36:05 AM] veribleVerilogLs language server client: couldn't create connection to server.
Launching server using command .vscode/verible-verilog-ls --indentation_spaces=4 failed. Error: spawn .vscode/verible-verilog-ls --indentation_spaces=4 ENOENT`

qwertycxz, 2024-10-26:
What @hi0t said is right. There are actually two same-name formatters. One is provided by the language server and another is provided by the formatter. If you manually select the formatter provided one, it works well. WHILE as their names are the same, VSCode cannot remember which formatter you want to use.
I have two ideas to solve this: one is remove the ls provided formatter. Or else we can just rename it.

PR #503 (2024-11-02): Fix glitches caused by Verible LS
Fix #449 
Fix #487 

Modified the option of `veribleVerilogLs` to include a `messageStrategy` that handles and modifies the language server capabilities by removing unnecessary diagnostic and formatting providers. 

#449 and #487 occurred as the same features are provided twice by *HDL support* and *verible LS*. All things work fine now.

Please notice that this patch does not disable any features of *Verible LS*. Instead, they are just suppressed and letting VSCode  not able to realize that the language server provide these features. VSCode therefore has to t [... cut]
closed 2024-11-02 by mshr-h