mshr-h/vscode-verilog-hdl-support #493: [BUG] A '\t' is inserted unexpectedly when instantiating verilog modules

RickyTino, 2024-08-07:
**Describe the bug**
Actually I am not quite sure if it is a bug or a feature. The plugin always insert a "\t" in front of the first signal connection when using the "module instantiation" provided by this plugin. 

**Environment (please complete the following information):**
 - OS: Linux version 3.10.0-1160.105.1.el7.x86_64 (mockbuild@kbuilder.bsys.centos.org) (gcc version 4.8.5 20150623 (Red Hat 4.8.5-44) (GCC) ) #1 SMP Thu Dec 7 15:39:45 UTC 2023
 - VS Code version 1.76.1
 - Extension version v1.11.3

**Steps to reproduce**
Steps to reproduce the behavior:
1. In certain source file press alt+i
2. Select any SV module to instantiate 
3. See a tab '\t' appear in front of the first signal connection

**Expected behavior**
No '\t' at all, or provide an option to set whether there be a '\t'

**Actual behavior**
[image]

**Additional context**
N/A

RickyTino, 2024-08-12:
It seems to be fix in v1.13.0. Not sure exactly which commit has taken effect. I diff-ed v1.13.0 and v1.11.1 but didn't figure out.

mshr-h, 2024-08-12:
Please include repro.

RickyTino, 2024-08-12:
> Please include repro.

Steps described above, but I have some new findings. latest update below:
(Let's assume this file indents using 4 spaces)
1. In certain .sv source file,  press 'tab' key to produce a 4-space indent;
2. press alt+i (or ctrl+shift+p, then choose "Verilog: Instantiate Module")
3. Select a .sv file to instantiate. This file should contain a module with parameter.
4. See a tab character '\t' appear in front of the first signal connection

OmarElfouly, 2024-10-13:
Hello @mshr-h  :)

I am new to contributing and I am excited to start with this issue. Can I ask if it would be preferred that the Instantiate Module not add a \t or would it be better to provide an option to decide whether or not to add a \t?

mshr-h, 2024-10-14:
@OmarElfouly Thank you for your comment!
I guess there's an option or status to indicate whether the indentation is a tab or spaces in vscode. So it would be great to follow it.

OmarElfouly, 2024-10-15:
Alright Ill start hunting it down and Ill create a PR as soon as Im done

PR #500 (2024-10-19): Perfered_instantiation_indent
resolves #493
closed 2024-10-19 by mshr-h